conservationtechlab/dencam #3: Integrate numeric focus evaluation tool into main dencam code

icr-ctl, 2021-02-22:


icr-ctl, 2022-01-04:
Now that Alex has made some great improvements to the utility that demonstrates this functionality, we should have BJ try it out and have him assess whether this tool would be a good thing to use in the field for focusing actual DenCams in situ.  If he thinks yes, then we should go ahead and integrate it into the appropriate page of the DenCam UI.

iamgarcia, 2022-01-11:
I've been working on integrating the focus score tool into the main dencam tool, but I am running into issues initializing a camera for the respective screen (ex. preview page = picamera, focus tool = opencv). This issue mostly has to do with the button handling being done using threading. If I try to initialize the cameras in the buttons class, there's no guarantee that the thread finished before the screen has updated, thus causing the application to crash because it's trying to access a camera that hasn't been initialized yet.

iingram, 2025-03-31:
This is is implemented but not reviewed for `lesehest`.  Given we are currently now using `fido` on deployment systems, we should implement this for that.  But if 2025 deployment data comes back well-focused when not using this tool maybe we drop the effort all together.