conservationtechlab/dencam #14: Placement of timestamp on frame

icr-ctl, 2021-11-24:
In some circumstances, we want the timestamp rendered onto the frame at the bottom of the image instead of the top.  It would be good to have a parameter in the configuration file that allows us to select whether timestamp is on top of frame or bottom.

iamgarcia, 2021-11-30:
Found this: https://picamera.readthedocs.io/en/release-1.13/faq.html#can-i-move-the-annotation-text

It appears that it's not possible to move the annotation text.

iamgarcia, 2021-12-28:
It appears that Picamera's firmware provides no means of moving the annotation text, according to the documentation found [here](https://buildmedia.readthedocs.org/media/pdf/picamera/latest/picamera.pdf) (Page 60, Section 5.12).