conservationtechlab/dencam #18: Screensaver on a new Pi is on and we need it off

icr-ctl, 2021-11-25:
A fresh Raspian install includes a screensaver which is on by default. It is easy to disable the screensaver through the desktop environment but we would really prefer to be able to integrate disabling the screensaver into a primary setup script (that we don't yet have) along with all/any other such steps so that a person setting up a DenCam never has to interact with the desktop environment and setup can be done by invoking one script.  Also would allow us to avoid having to add instructions for disabling the screensaver via the desktop environment into the README.

The question becomes how does one disable the screensaver using the CLI?

iamgarcia, 2021-12-01:
I looked into this matter and found a method that works. I tested the method on my Raspberry Pi and I can vouch that it does disable the screensaver and screen blanking. Having said that, I should note that the GUI doesn't reflect the changes made in regards to the disabled screensaver and screen blanking.

This is the forum that I referred to: https://raspberrypi.stackexchange.com/questions/2059/disable-screen-blanking-in-x-windows-on-raspbian#2079

iamgarcia, 2022-01-11:
I noticed in the readme file that there are different usages of the pip and python command (ex. pip or pip3, python or python3). Which version should I stick to when writing the terminal commands for the install.sh script?

icr-ctl, 2022-01-11:
Within the virtual environments we recommend (and explain the setup of) we're assuming python3 so sticking within pip and python should be fine if the user is currently within the VE but it may be better to be explicit with pip3 and python3.

Note: README also calls for 
    python setup.py install
but this is deprecated now so we should use `pip install` for this, too.

closed 2022-01-14 by icr-ctl