conservationtechlab/dencam #21: Power Data Log from SunSaver MPPT via ModBus

bkirshh, 2021-12-01:
Power Data Log

Modbus Values to pull at 60 min intervals to a log file co-located with the video data. This log file could sit in the same directory as all the daily video data folders (/media/pi/external_storage). Each log entry should have the Pi's hardware clock value associated with it as the MPPT charge controller does not have an accurate clock. 

Pi Hardware Clock Value
SunSaver MPPT Battery Voltage (V)	
SunSaver MPPT Array Voltage (V)	
SunSaver MPPT Load Voltage (V)	
SunSaver MPPT Charge Current (A)	
SunSaver MPPT Load Current (A)	
SunSaver MPPT RTS Temperature (deg C)
SunSaver MPPT Ambient Temperature 
MPPT Temp	
SunSaver MPPT Charge State ()	
SunSaver MPPT Ah Charge(Daily) (Ah)	
SunSaver MPPT Ah Load(Daily) (Ah)	
SunSaver MPPT Alarm (Daily)

Modbus Values to pull at 12 hour intervals to a log file in ~/dencam. Each log entry should have the Pi's hardware clock value associated with it as the MPPT charge controller does not have an accurate clock. 

Pi Hardware Clock Value
SunSaver MPPT Battery Voltage (V)	
SunSaver MPPT Array Voltage (V)	
SunSaver MPPT Load Voltage (V)	
SunSaver MPPT Charge Current (A)	
SunSaver MPPT Load Current (A)	
SunSaver MPPT RTS Temperature (deg C)
SunSaver MPPT Ambient Temperature 
MPPT Temp	
SunSaver MPPT Charge State ()	
SunSaver MPPT Ah Charge(Daily) (Ah)	
SunSaver MPPT Ah Load(Daily) (Ah)	
SunSaver MPPT Alarm (Daily)

EliasMartinezJ, 2021-12-09:
Hey BJ, 
   i was wondering what information you would like being displayed on the Solar display page.

bkirshh, 2021-12-10:
Hi Elias,

Thanks for the followup and your efforts. I think the items below would be sufficient. I have placed this request in "Add display page for reporting solar charge controller information #17” on the Issues tab if that helps. 

BJ

Power Status Screen

I would suggest this page within dencam.py would allow operators to check power related values captured from the MPPT Charge Controller via ModBus. The values below would be helpful within the power status page...

Desired Values to display
SunSaver MPPT Battery Voltage (V)
SunSaver MPPT Array Voltage (V)
SunSaver MPPT Charge Current (A)
SunSaver MPPT Load Current (A)

Optional if additional space is available
SunSaver MPPT Ah Charge(Daily) (Ah)
SunSaver MPPT Ah Load(Daily) (Ah)

EliasMartinezJ, 2021-12-16:
Hi BJ, 
 Was just wondering if it would be acceptable to store the hourly MPPT values into one giant csv file rather than having it separated by days

Elias

bkirshh, 2021-12-20:
Hi Elias,

One file works for me.

BJ

EliasMartinezJ, 2022-01-06:
Hi BJ,

 You said you were using a cronjob to reboot the dencam each day at 1am, Im updating the readme right now to include the cronjobs we have set up for the SunSaver and was hoping you could send over the comman you're using for that cronjob so i could include it while im at it.

Elias

bkirshh, 2022-01-06:
Hi Elias,

Here is how I am doing the reboot and the hardware clock sync commands….

$ sudo nano /etc/crontab

@reboot         root    hwclock --hctosys --utc
30 *    * * *   root    hwclock --hctosys --utc
0  1    * * *   root    /sbin/shutdown -r now
#