conservationtechlab/dencam #32: Ghost Drive

bkirshh, 2022-01-20:
On occasion when the power is abruptly cut to the RaspberryPi running dencam, USB attached storage drives (microSD cards in card readers) do not appear to be properly ejected. The name of the drive appears to stick in /media/pi. 

Several of my Pi's (I am currently testing five with four different batteries) have been neglecting to display the second status page. This, as I have come to find out, is because the system is remembering the names of my attached storage when the pi is not properly shut down (nearly every time I just pull the plug). This is easy enough to solve (just ssh in and cd /media/pi and sudo rmdir X). But in the field, this could be catastrophic as these pseudo-replicated drives prevent dencam.py from writing to either the external or internal storage.

iamgarcia, 2022-02-04:
@bkirshh I have been unsuccessful in being able to replicate this issue. Could you provide me with the exact steps that you did to cause the ghost drive issue? Thanks!

bkirshh, 2022-02-09:
After my devices stabilize (we fix the slow DenCam issue) I'll try again to replicate. In the meantime can you try pulling the power on a functioning DenCam over and over again, maybe ten times. Don't properly shut it down, just pull the power as the files are writing. See what happens.

iingram, 2025-02-27:
There was recent evidence that this issue was rearing its ugly head.  Alex's fix from way back when that he was unable to truly test because the issue couldn't be recreated was cherry-picked and tested to the degree it could be and incorporated into `v0.0.7` because it was deemed the risks incurred by not doing so outweighed the risks incurred by doing so.  We should try to do some more testing to both verify the fix really addresses all the ways the problem can manifest and also that there is no regression.

closed 2025-03-06 by iingram