conservationtechlab/dencam #53: MODBUS and Meterbus Addresses

bkirshh, 2022-02-09:
To sum things up, only 1 of 5 Pis was "seeing" the SunSaver MPPT device. I had a hunch and did some digging and found that the MODBUS address needs to be set to 1. Using Morningstar Solar's software titled MSView, we can configure this addresses.

I can type a full description of how to do this later unless we find a workaround like a MODBUS scan for devices upon initiation of the software.

icr-ctl, 2022-02-09:
Nice detective work!  

SDZWA team is this something we knew about at one point or another but didn't capture into the install/setup process or is this fresh news? @tkswanson and @EliasMartinezJ: I imagine this had to be done to our SunSaver at some point, right?

bkirshh, 2022-02-09:
If nobody touched this, It could have been luck of the draw that you happened to have a device assigned to MODBUS 1 at the factory.