conservationtechlab/dencam #54: Slow DenCam/Slow Pis

bkirshh, 2022-02-09:
DenCam is running very slow on my Raspberry Pi 3B+s, and some (3/5 devices) have stopped recording video. The slowness is enough that when you press a button to advance the pages, many seconds elapse before the next page appears. In some cases, the pages do not advance at all. Sometimes, the recording and SunSaver pages do not appear, but the camera preview screen will. No wlan data on page 1 typically precede this event but not always. I'll upload a video in Slack. 

If I toggle "off" recording, speeds typically increase when flipping through the pages. 

To help troubleshoot, I am changing the logging interval for sunsaver_log.pi to once an hour in crontab (1 * * * *). I do realize that we will need to depress the "action" button to update the SunSaver data in DenCam when using infrequent logging intervals.

icr-ctl, 2022-02-09:
@EliasMartinezJ and @iamgarcia I have assigned this to both of you because clearly this is a big issue, it is possible you'll have to confer on things you each have been doing to solve it, and it is probably just better to have two heads on it.

icr-ctl, 2022-02-09:
> DenCam is running very slow on my Raspberry Pi 3B+s, and some (3/5 devices) have stopped recording video. The slowness is enough that when you press a button to advance the pages, many seconds [...]

@bkirshh to confirm: I believe it was our understanding that the SunSaver logging cronjob **would** be once per hour in the deployments. Were you thinking it would be more frequent?

bkirshh, 2022-02-09:
When I set this up with Elias, we set things for 1/min. Now are are logging 1/hour. 1/hour will be plenty of data.

bkirshh, 2022-02-09:
@EliasMartinezJ and @iamgarcia let me know if you want to setup a call to workshop this problem. Thanks in advance for your help on this.

iingram, 2025-03-31:
We think if this issue exists at all it is the same issue that is now captured in other issues, particularly #79, so we're retiring this one.

closed 2025-03-31 by iingram