conservationtechlab/dencam #78: Updates for Raspberry Pi 5/Bookwork

sumi1194, 2024-08-22:
Creating issue to capture change points necessary for upgrading to Raspberry Pi 5/Bookworm

sumi1194, 2024-08-22:
**Setting up virtual environments**

Bookworm requires a virtual environment for pip installation packages. `sudo pip` is no longer allowed. The file structure is also different so set up for virtual environments is slightly changed.

Install virtual environment packages 
`sudo apt install virtualenv virtualenvwrapper`

Add to bash file: 
```
echo -e "\n# Virtual environment setup" >> ~.bashrc
echo "export WORKON_HOME=~/.virtualenvs" >> ~.bashrc
echo "export VIRTUALENVWRAPPER_PYTHON=/usr/bin/python3" >> ~.bashrc
echo "source /usr/share/virtualenvwrapper.sh" >> ~.bashrc
source ~.bashrc
```

Restart terminal

Create new virtualenv:
`mkvirtualenv --system-site-packages <venv>`

This is espeically important if you are using libcamera (picamera), since it is not yet on pypi

sumi1194, 2024-08-22:
**GPIO Pins**

RPI.GPIO and wiringpi are no longer supported on raspberry pi 5. The recommended GPIO package is gpiozero. 
Documentation: [https://gpiozero.readthedocs.io/en/stable/index.html]

iingram, 2025-03-31:
This information needs to be captured into the README and then we can retire this issue.