conservationtechlab/dencam #105: dencam.py needs to accept different camera types as an input

kgarwoodsdzwa, 2024-10-29:
if we are offering 3 versions of cameras (ptz, picamera1 and picamera2) dencam.py will need to be able to differentiate to choose which recorder.py classes to use. the solution could be a change to the config file where dencam.py can read in the camera type to choose which errors to handle and which of those classes to call

iingram, 2024-10-31:
Per our conversation yesterday, I think we return to the three separate high-level application modules, one for each hardware target, so that we don't have to do conditional imports.  What we should do, though, in a related vein, is figure out ways to further encapsulate mechanics common between these three modules.

iingram, 2024-12-04:
Close but waiting till we've merged Mimir work and renamed `dencam.py` to `fenrir.py` before closing the issue.