conservationtechlab/dencam #126: Get image prepared in #125 to Elbert and Christian

iingram, 2025-01-29:
We want to get `lesehest` to Christian as soon as we can so he can start testing with it. Per #125 the plan is to prepare an image and send it to him somehow so he can flash it onto an SD card he has in Longyearbyen and try it on one of the Fenrir units he has there.  (Worth noting Elbert would need an image that runs on Pi4 or Pi5)

kgarwoodsdzwa, 2025-01-29:
Elbert has actually been following my lesehest readme and has it installed already, letting me know any issues as he goes. So I probably won't make a bootable image for 4b, just 3b for christian unless there is another reason to make one for a 4b

kgarwoodsdzwa, 2025-02-04:
Sent a box link that contains the .img and an instruction document for how to install it. Once Christian is able to try it we can see if there are issues.

closed 2025-02-12 by iingram